CJT2325/CameraView #1: Err when MediaRecorder.start

FanaticLLL, 2017-02-24:
java.lang.RuntimeException: start failed.
                      at android.media.MediaRecorder._start(Native Method)
                      at android.media.MediaRecorder.start(MediaRecorder.java:914)
                      at com.cjt2325.cameralibrary.JCameraView.startRecord(JCameraView.java:401)
                      at com.cjt2325.cameralibrary.JCameraView.access$1000(JCameraView.java:31)
                      at com.cjt2325.cameralibrary.JCameraView$1.record(JCameraView.java:132)
                      at com.cjt2325.cameralibrary.CaptureButton$RecordRunnable.run(CaptureButton.java:357)
                      at android.os.Handler.handleCallback(Handler.java:739)
                      at android.os.Handler.dispatchMessage(Handler.java:95)
                      at android.os.Looper.loop(Looper.java:145)
                      at android.app.ActivityThread.main(ActivityThread.java:6918)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at java.lang.reflect.Method.invoke(Method.java:372)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199)

CJT2325, 2017-02-24:
I have fixed this BUG , you try this "cjt.library.wheel:camera:0.0.7"

FanaticLLL, 2017-02-24:
👍

发自网易邮箱大师
On 02/24/2017 18:14, Mc_JT wrote:

I have fixed this BUG , you try this "cjt.library.wheel📷0.0.7"

closed 2017-05-04 by CJT2325