CJT2325/CameraView #2: 摄像头切换会闪退

wuchao226, 2017-03-01:
点击切换摄像头时，程序崩溃

hesshzy, 2017-03-01:
切换前置摄像头时闪退。
小米4LTE，Android6.0

03-01 11:49:40.187 32101-32101/com.cjt2325.cameraview D/AndroidRuntime: Shutting down VM
03-01 11:49:40.193 32101-32101/com.cjt2325.cameraview E/AndroidRuntime: FATAL EXCEPTION: main
                                                                        Process: com.cjt2325.cameraview, PID: 32101
                                                                        java.lang.RuntimeException: setParameters failed
                                                                            at android.hardware.Camera.native_setParameters(Native Method)
                                                                            at android.hardware.Camera.setParameters(Camera.java:2011)
                                                                            at com.cjt2325.cameralibrary.JCameraView.setStartPreview(JCameraView.java:281)
                                                                            at com.cjt2325.cameralibrary.JCameraView.access$700(JCameraView.java:35)
                                                                            at com.cjt2325.cameralibrary.JCameraView$2.onClick(JCameraView.java:227)
                                                                            at android.view.View.performClick(View.java:5207)
                                                                            at android.view.View$PerformClick.run(View.java:21177)
                                                                            at android.os.Handler.handleCallback(Handler.java:739)
                                                                            at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                            at android.os.Looper.loop(Looper.java:148)
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5438)
                                                                            at java.lang.reflect.Method.invoke(Native Method)
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:739)
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:629)
03-01 11:49:40.210 32101-32101/com.cjt2325.cameraview I/Process: Sending signal. PID: 32101 SIG: 9

CJT2325, 2017-03-01:
已经修复这个BUG，你使用0.1.1版本试试，因为我没有多台实体机可以测试，所以有的BUG可能无法测到

hesshzy, 2017-03-01:
更新到0.1.1版本后，这个bug在小米4机型上已修复

wuchao226, 2017-03-01:
更新到0.1.1版本后，这个bug在三星s7edge上依然存在

CJT2325, 2017-03-01:
@wuchao226 你用0.1.2版本试试，我修改了前置，后置和录像的分辨率，我手头上的实体机都成功运行。

karmalove, 2017-03-03:
华为荣耀5c切换前置摄像头也是崩溃（0.2.2版本）

CJT2325, 2017-03-03:
@karmalove 你贴一下异常给我看看，或者可以加一下QQ吗？

CJT2325, 2017-03-04:
@karmalove 三星s7edge前置摄像头切换崩溃已经修复，你试试你的机型能否运行

karmalove, 2017-03-06:
我qq：947154640

closed 2017-05-04 by CJT2325