CJT2325/CameraView #3: 更新到0.1.2版本后，三星s7edge拍摄视频和切换前置摄像头时奔溃

wuchao226, 2017-03-02:


wuchao226, 2017-03-03:
录制视频崩溃bug
03-03 17:49:16.743 11845-19237/? E/MediaRecorderService: Request requires android.permission.RECORD_AUDIO
03-03 17:49:16.743 4381-4381/com.xdy.qxzst.erp D/AndroidRuntime: Shutting down VM
03-03 17:49:16.763 1012-13268/? D/mm-camera-CORE: TSAec_process works: stats' frame_id=179, 0, 0
03-03 17:49:16.773 1012-13268/? I/mm-camera-sensor: sensor_apply_exposure:946s_gain=0x1000168, l_gain=0x1000168, s_linecount=2876,l_linecount=2876, fl_lines=2904
03-03 17:49:16.773 11992-12979/? I/EDMNativeHelperService: isCameraEnabled
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err: java.lang.RuntimeException: setAudioSource failed.
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.media.MediaRecorder._setAudioSource(Native Method)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.media.MediaRecorder.setAudioSource(MediaRecorder.java:488)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView.startRecord(JCameraView.java:498)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView.access$900(JCameraView.java:39)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView$1.record(JCameraView.java:152)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.CaptureButton$RecordRunnable.run(CaptureButton.java:349)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.os.Looper.loop(Looper.java:158)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:7233)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
03-03 17:49:16.793 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err: java.lang.RuntimeException: setAudioSource failed.
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.media.MediaRecorder._setAudioSource(Native Method)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.media.MediaRecorder.setAudioSource(MediaRecorder.java:488)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView.startRecord(JCameraView.java:498)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView.access$900(JCameraView.java:39)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView$1.record(JCameraView.java:152)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.CaptureButton$RecordRunnable.run(CaptureButton.java:349)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.os.Looper.loop(Looper.java:158)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:7233)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
03-03 17:49:16.803 4381-4381/com.xdy.qxzst.erp W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
03-03 17:49:16.813 1012-13268/? I/mm-camera-sensor: sensor_apply_exposure:946s_gain=0x100016B, l_gain=0x100016B, s_linecount=2876,l_linecount=2876, fl_lines=2904

wuchao226, 2017-03-03:
切换摄像头bug
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err: java.lang.RuntimeException: setParameters failed
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at android.hardware.Camera.native_setParameters(Native Method)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at android.hardware.Camera.setParameters(Camera.java:1984)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView.setStartPreview(JCameraView.java:342)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView.access$600(JCameraView.java:39)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at com.cjt2325.cameralibrary.JCameraView$2.onClick(JCameraView.java:243)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at android.view.View.performClick(View.java:5697)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at android.view.View$PerformClick.run(View.java:22526)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at android.os.Looper.loop(Looper.java:158)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:7233)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
03-03 17:55:42.873 13625-13625/com.xdy.qxzst.erp W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
03-03 17:55:42.883 13625-13625/com.xdy.

CJT2325, 2017-03-03:
给个Q，上Q聊 @wuchao226

wuchao226, 2017-03-03:
@CJT2325 1163719064

closed 2017-05-04 by CJT2325