CJT2325/CameraView #14: 录像后预览的时候不全屏，然后点cancel按钮返回预览界面也不再全屏了

djun100, 2017-04-06:


CJT2325, 2017-04-06:
```
//你可以这些添加一下沉浸式模式来设置全屏
public class MainActivity extends AppCompatActivity {
    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
    }
    @Override
    public void onWindowFocusChanged(boolean hasFocus) {
        super.onWindowFocusChanged(hasFocus);
        if (hasFocus && Build.VERSION.SDK_INT >= 19) {
            View decorView = getWindow().getDecorView();
            decorView.setSystemUiVisibility(
                View.SYSTEM_UI_FLAG_LAYOUT_STABLE
                | View.SYSTEM_UI_FLAG_LAYOUT_HIDE_NAVIGATION
                | View.SYSTEM_UI_FLAG_LAYOUT_FULLSCREEN
                | View.SYSTEM_UI_FLAG_HIDE_NAVIGATION
                | View.SYSTEM_UI_FLAG_FULLSCREEN
                | View.SYSTEM_UI_FLAG_IMMERSIVE_STICKY);
        }
    }
}
```

ZSFeng, 2017-04-28:
1、现在1.0.0版本的Demo也是同样的问题，录完视频，预览的时候不全屏。
2、录完视频，预览的时候点击取消，然后再点击三角尖按钮退出会报这么一个错：
E/WindowManager: Activity com.cjt2325.cameraview.MainActivity has leaked window com.android.internal.policy.impl.PhoneWindow$DecorView{41acedb8 V.E..... R.....I. 0,0-684,242} that was originally added here
android.view.WindowLeaked: Activity com.cjt2325.cameraview.MainActivity has leaked window com.android.internal.policy.impl.PhoneWindow$DecorView{41acedb8 V.E..... R.....I. 0,0-684,242} that was originally added here
at android.view.ViewRootImpl.<init>(ViewRootImpl.java:424)
at android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:218)
at android.view.WindowManagerImpl.addView(WindowManagerImpl.java:69)
at android.app.Dialog.show(Dialog.java:281)
at android.app.AlertDialog$Builder.show(AlertDialog.java:951)
at android.widget.VideoView$4.onError(VideoView.java:507)
at android.media.MediaPlayer$EventHandler.handleMessage(MediaPlayer.java:2153)
at android.os.Handler.dispatchMessage(Handler.java:107)
at android.os.Looper.loop(Looper.java:194)
at android.app.ActivityThread.main(ActivityThread.java:5543)
at java.lang.reflect.Method.invokeNative(Native Method)
at java.lang.reflect.Method.invoke(Method.java:525)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:835)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:602)
at dalvik.system.NativeStart.main(Native Method)

CJT2325, 2017-04-28:
@ZSFeng 你是什么机型

CJT2325, 2017-04-28:
@ZSFeng 你试试 1.0.1

ZSFeng, 2017-04-28:
@CJT2325 大可乐DKL01    Android 4.2.2

ZSFeng, 2017-04-28:
@CJT2325 1.0.1测试版本不能用，一按住录像，画面就卡住了，它就当拍照，但是那个录像进度还在走...

CJT2325, 2017-04-28:
应该有的机型不适配，我华为魅族都行

closed 2017-05-05 by CJT2325