CJT2325/CameraView #18: 录制视频时候，时间过短，会崩。请问怎么解决

Zhangxd77, 2017-04-12:
java.lang.RuntimeException: stop failed.              
android.media.MediaRecorder.native_stop(Native Method)
崩溃日志如上！网上找说加mediaRecorder.setOnErrorListener(null);这些都无效。请问知道解决办法吗？
qq:285601517.麻烦知道的和我联系。谢谢

huaf22, 2017-04-27:
我也遇到同样的问题, 你有解决这个问题吗? 
下面是我的错误日志:
```
E/MediaRecorder: stop failed: -1007
java.lang.RuntimeException: stop failed.
    at android.media.MediaRecorder.stop(Native Method)
    at cn.mobile.tv.service.VideoRecordService.stopRecord(VideoRecordService.java:128)
    at cn.mobile.tv.service.VideoRecordService$1.run(VideoRecordService.java:111)
    at android.os.Handler.handleCallback(Handler.java:739)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:135)
    at android.os.HandlerThread.run(HandlerThread.java:61)
```

CJT2325, 2017-04-27:
试试1.0.0最新的版本

CJT2325, 2017-04-27:
@huaf22

huaf22, 2017-04-27:
@CJT2325  那这个问题具体在代码中是是怎么解决的? 需要设置什么参数?

CJT2325, 2017-04-27:
你可以跳转stop这里的源码看他上面注释，说开启录像后立刻stop会直接报错，所以我就在立刻stop的时候，用一个线程延迟操作一下stop

huaf22, 2017-04-27:
@CJT2325  我看到延迟操作的代码了, 但是我现在的问题是开始录制10s后停止录制也会出现这样的问题,  保存视频的文件也不能播放

CJT2325, 2017-04-27:
QQ 445263848

closed 2017-05-04 by CJT2325