CJT2325/CameraView #22: 手机横着拍完照片或录完视频，预览的时候，是竖着显示

ZSFeng, 2017-04-27:
请问如何在预览的时候也是横着的

CJT2325, 2017-04-27:
这个还没有实现····

ZSFeng, 2017-04-28:
要是想实现有什么思路吗？求大神解答，不然横着拍出来的效果不好...

CJT2325, 2017-04-28:
拍照倒不是很难，不过录视频就可能麻烦很多

ZSFeng, 2017-04-28:
那拍照能分析下如何做吗？希望以后能实现这个功能

CJT2325, 2017-05-02:
@ZSFeng 已经添加该功能，拍照和视频后的浏览都会根据手机水平方向进行相应旋转

ZSFeng, 2017-05-02:
@CJT2325   1.0.2版本，测试demo有几个问题：
1、竖着录像后预览没有之前版本的全屏，好像被压瘪的感觉？
2、前后摄像头切换，视图好像有点变形
3、录像比较模糊的感觉
4、拍照录像后预览，还能点击聚焦

ZSFeng, 2017-05-03:
@CJT2325 上面有一些问题可能是不同手机适配问题。然后有些手机一按住录像，界面视图就卡住了，但进度条还在走，后台log：
E/MediaPlayer: error (100, 0)
I/CameraFramework: handleMessage: 1
E/CameraFramework: Error 100
放开log：
E/MediaPlayer: Unable to to create media player
W/System.err: java.io.IOException: setDataSourceFD failed.: status=0x80000000
W/System.err:     at android.media.MediaPlayer.setDataSource(Native Method)
W/System.err:     at android.media.MediaPlayer.setDataSource(MediaPlayer.java:1032)
W/System.err:     at android.media.MediaPlayer.setDataSource(MediaPlayer.java:1011)
W/System.err:     at android.media.MediaPlayer.setDataSource(MediaPlayer.java:974)
W/System.err:     at com.cjt2325.cameralibrary.JCameraView$2$3$1.run(JCameraView.java:281)
W/System.err:     at java.lang.Thread.run(Thread.java:838)

CJT2325, 2017-05-03:
445263848 Q

closed 2017-05-04 by CJT2325