CJT2325/CameraView #27: 第一次启动出现界面出现这样的问题。

guowang1102, 2017-05-06:
FATAL EXCEPTION: Thread-5
                                                            
      java.lang.RuntimeException: Fail to connect to camera service
                      at android.hardware.Camera.<init>(Camera.java:692)
                      at android.hardware.Camera.open(Camera.java:488)

CJT2325, 2017-05-06:
你是下载我的Demo apk的吗？

guowang1102, 2017-05-06:
集成到项目中，startactivity打开。

guowang1102, 2017-05-06:
只有第一次授权完之后界面异常退出报这个错。再次进去就完全正常了

CJT2325, 2017-05-06:
你的设备是不是6.0以上的吗？如果是有可能是我的Demo的动态获取权限有点问题

guowang1102, 2017-05-06:
是7.1的系统

CJT2325, 2017-05-06:
那有可能是我动态获取权限的问题，因为我那部分代码也是简单复制粘贴而已，你可以自己添加一下动态权限获取试试看

ddssingsong, 2017-05-07:
这个是因为退出之后没有及时释放摄像头资源才引起的，我测试了一天

CJT2325, 2017-05-07:
难道是要在surfaceview回调接口的destroy再释放一次资源吗？

ddssingsong, 2017-05-07:
是这样的，当你权限没有给与的时候，进行初始化操作或崩溃，崩溃之后硬件一直被占用，下次进入的时候进不到open的方法，只有重启手机在给与权限才能打开。。。我是使用debug跳过内容才知道的

guowang1102, 2017-05-07:
是的，就是授权的时候，也就是第一次会出现这情况，有没有改进的方法。

closed 2017-05-26 by CJT2325

chenjunyubt, 2018-11-11:
这问题解决了吗？  新下载的 1.1.9  也还有这个问题， 允许权限后 一直黑屏， 第二次进入才正常