CJT2325/CameraView #38: 长按摄像，弹出录音权限提示！！！动画会自动执行完成

missu263, 2017-05-22:
在真机ViVo Y66 android6.0上，首次运行，长按摄像的时候，会弹出录音权限提示框，这时候提示框会覆盖界面的最上层啊，然后点击授权，松开了手指，动画也会执行完duration。
当我去选择权限时，控件无法捕获ACTION_DOWN、ACTION_MOVE、ACTION_UP和ACTION_CANCEL事件，不能及时消除动画效果。
请问大神，这个有没有办法去解决啊。。。

CJT2325, 2017-05-22:
应该可以现在onTouch先判断一下有无获取权限

CJT2325, 2017-05-22:
你的手机是6.0的，第一次打开的时候没有弹出权限获取提示框嘛？

missu263, 2017-05-22:
我手头上的测试机ViVo Y66 android6.0，第一次打开会自动弹出拍摄的权限提示框，但是录音的提示框要按下按钮之后，开始录制的时候才会弹出这个录音的权限哦。。。
应该是提示框的Dialog弹出，界面控件失去焦点了，所以captureButton无法再去捕获MotionEvent，所以动画默认自动全部执行完毕了。。。
关于您说的在操作之前去检查权限，这个在ViVo Y66 android6.0上，是不生效的哦。
所以想请教您，能否在自定义控件captureButton上去做相关的处理。比如判断有没有弹出框，这样去取消动画了。
以上，非常感谢您的回复 ：）

CJT2325, 2017-05-22:
你试试看在 MOTION_CANCEL 这个onTouch 事件打印一下LOG，看看弹出Dialog时候并且点击后会不会执行这个事件，如果执行的话你就可以在这里去停止动画

closed 2017-05-26 by CJT2325