CJT2325/CameraView #40: 大神我测出来一个bug

zhangxiaobin123456, 2017-05-24:
在录制视频的时候，频繁的按住松开会提示录制时间过短，几次之后就报如下error，不知道是不是我手机的问题。
 Process: com.example.gridviewtest, PID: 3948
                                                                        java.lang.RuntimeException: stop failed.
                                                                            at android.media.MediaRecorder.stop(Native Method)
                                                                            at com.cjt2325.cameralibrary.CameraInterface.stopRecord(CameraInterface.java:311)
                                                                            at com.cjt2325.cameralibrary.JCameraView$2$2.run(JCameraView.java:245)
                                                                            at android.os.Handler.handleCallback(Handler.java:733)
                                                                            at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                            at android.os.Looper.loop(Looper.java:136)
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5019)
                                                                            at java.lang.reflect.Method.invokeNative(Native Method)
                                                                            at java.lang.reflect.Method.invoke(Method.java:515)
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:779)
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:595)
                                                                            at dalvik.system.NativeStart.main(Native Method)

CJT2325, 2017-05-24:
你好，这个BUG不是你设备的原因，是这个项目的存在BUG
This happens if stop() is called immediately after start()；
这是MediaRecorder接口stop()方法的说明，如果在star()之后立刻stop()就会抛出异常，你现在看到的这份代码我的处理是如果录制视频过短，会延迟1000毫秒再去stop()，但是我没有处理频繁按住松开这部分逻辑，在我本地我已经处理了，不过逻辑有点不太清晰，等完善之后我更新上去。

zhangxiaobin123456, 2017-05-24:
我找到源码是试着将 CameraInterface类中的 stopRecord方法 中 try{}catch{}里边的 IllegalStateException 改成了 Exception , 
 if (mediaRecorder != null && isRecorder) {
            try {
                mediaRecorder.stop();
                isRecorder = false;
            } catch (Exception e) {
                e.printStackTrace();
            }

点了点暂时好像没出bug

CJT2325, 2017-05-24:
嗯，可以尝试一下，然后你最好在catch到异常后将资源全部释放掉，或者放在finily里面也可以

zhangxiaobin123456, 2017-05-24:
等于就是将  stopRecord方法剩下的代码都放到了finally 里边吧

CJT2325, 2017-05-24:
嗯，就是一些状态的更改，和资源的释放

zhangxiaobin123456, 2017-05-24:
谢谢大神指点，我放进去了，试着频繁操作并没有崩溃

CJT2325, 2017-05-24:
:thumbsup: 我找个时间用你的方法处理一下

closed 2017-05-26 by CJT2325