CJT2325/CameraView #43: 触摸对焦代码

lxxself, 2017-05-25:
```
private static Rect calculateTapArea(float x, float y, float coefficient, Camera.Size previewSize) {
        float focusAreaSize = 300;
        int areaSize = Float.valueOf(focusAreaSize * coefficient).intValue();
        int centerX = (int) (x / previewSize.width - 1000);
        int centerY = (int) (y / previewSize.height - 1000);

        int left = clamp(centerX - areaSize / 2, -1000, 1000);
        int top = clamp(centerY - areaSize / 2, -1000, 1000);

        RectF rectF = new RectF(left, top, left + areaSize, top + areaSize);

        return new Rect(Math.round(rectF.left), Math.round(rectF.top), Math.round(rectF.right), Math.round(rectF
                .bottom));
    }
```
你好，这一部分代码  int centerX = (int) (x / previewSize.width - 1000);不需要加*2000么？即  int centerX = (int) (x / previewSize.width *2000 - 1000);

CJT2325, 2017-05-25:
嗯，需要添加，感谢提出，我修改后更新上去

CJT2325, 2017-05-25:
我更新了代码，你拉取最新代码试试看，我把previewSize改为使用设备的长宽

closed 2017-05-26 by CJT2325