CJT2325/CameraView #45: 在 7.0 系统上会崩溃

coolyunyun, 2017-05-31:
在 7.0 系统上会崩溃  不妨测试下。

CJT2325, 2017-05-31:
是权限问题吗

CJT2325, 2017-05-31:
你报一下你的设备

coolyunyun, 2017-05-31:
我下载那个demo apk 也是报错啊   你里面没加权限判断吗？  类库里
设备  华为的7.0设备

CJT2325, 2017-05-31:
嗯，判断了，你贴一下log看看报错原因，我这边7.0机子能运行

coolyunyun, 2017-05-31:
FATAL EXCEPTION: main
                                                              Process: com.tuoying.zy, PID: 4075
                                                              java.lang.RuntimeException: takePicture failed, error=-38
                                                                  at android.hardware.Camera.native_takePicture(Native Method)
                                                                  at android.hardware.Camera.takePicture(Camera.java:1772)
                                                                  at android.hardware.Camera.takePicture(Camera.java:1705)
                                                                  at com.cjt2325.cameralibrary.CameraInterface.takePicture(CameraInterface.java:366)
                                                                  at com.cjt2325.cameralibrary.JCameraView$2.takePictures(JCameraView.java:202)
                                                                  at com.cjt2325.cameralibrary.CaptureLayout$2.takePictures(CaptureLayout.java:160)
                                                                  at com.cjt2325.cameralibrary.CaptureButton.handlerUnpressByState(CaptureButton.java:194)
                                                                  at com.cjt2325.cameralibrary.CaptureButton.onTouchEvent(CaptureButton.java:178)
                                                                  at android.view.View.dispatchTouchEvent(View.java:10012)
                                                                  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2671)
                                                                  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2358)
                                                                  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2671)
                                                                  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2358)
                                                                  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2671)
                                                                  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2358)
                                                                  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2671)
                                                                  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2358)
                                                                  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2671)
                                                                  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2358)
                                                                  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2671)
                                                                  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2358)
                                                                  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2671)
                                                                  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2358)
                                                                  at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2671)
                                                                  at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2358)
                                                                  at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:447)
                                                                  at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1871)
                                                                  at android.app.Activity.dispatchTouchEvent(Activity.java:3213)
                                                                  at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                                                                  at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:409)
                                                                  at android.view.View.dispatchPointerEvent(View.java:10232)
                                                                  at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4856)
                                                                  at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4717)
                                                                  at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4249)
                                                                  at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4302)
                                                                  at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4268)
                                                                  at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4395)
[... cut ...]
                                                                  at android.app.ActivityThread.main(ActivityThread.java:6523)
                                                                  at java.lang.reflect.Method.invoke(Native Method)
                                                                  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:941)
                                                                  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:831)

CJT2325, 2017-05-31:
加QQ 445263848

closed 2017-06-07 by CJT2325

jiugege, 2018-03-27:
我也出现了这个问题，手机是乐视6.0系统，请问下这个问题修复了吗，求教大神