CJT2325/CameraView #46: 在Android5.1及以下版本的手机上无法预览

sanwuwy, 2017-05-31:
我测试使用的手机是魅蓝3(android5.1)和华为荣耀PE-CL00(android4.4.2)
点击进入之后后置摄像头的预览画面是黑屏的，需要调用一下切换摄像头按钮，切换成前置摄像头才可以看到前置摄像头的预览画面，再切换回后置摄像头，也可以看到后置摄像头的预览画面了。

CJT2325, 2017-05-31:
你再下载试试看，这个问题只是单纯我的demo权限获取问题，因为要处理6.0以上的，忘记兼容下来··

CJT2325, 2017-05-31:
我更新了demo了   你重新下载apk试试

sanwuwy, 2017-05-31:
我重新下载apk试了，在魅蓝3的手机上可以了。
但是在华为荣耀PE-CL00(android4.4.2)手机上，以下两种情况下还是会复现：
1. 如果在APP预览界面将APP强制杀死，再次进入APP，预览页面还是黑屏的，复现概率高。
2. 多次快速进入退出APP，预览界面偶尔也会黑屏，复现概率低。
另外华为手机在预览界面为黑屏时偶尔还会报错：
05-31 19:28:47.620 19537-19537/com.zxing.cameraapplication E/InputEventReceiver: Exception dispatching input event.
05-31 19:28:47.620 19537-19537/com.zxing.cameraapplication E/MessageQueue-JNI: Exception in MessageQueue callback: handleReceiveCallback
05-31 19:28:47.635 19537-19537/com.zxing.cameraapplication E/MessageQueue-JNI: java.lang.RuntimeException: autoFocus failed
                                                                                   at android.hardware.Camera.native_autoFocus(Native Method)
                                                                                   at android.hardware.Camera.autoFocus(Camera.java:1148)
                                                                                   at com.cjt2325.cameralibrary.CameraInterface.handleFocus(CameraInterface.java:532)
                                                                                   at com.cjt2325.cameralibrary.JCameraView.setFocusViewWidthAnimation(JCameraView.java:470)
                                                                                   at com.cjt2325.cameralibrary.JCameraView.onTouchEvent(JCameraView.java:408)
                                                                                   at android.view.View.dispatchTouchEvent(View.java:7723)
                                                                                   at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2212)
                                                                                   at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1945)
                                                                                   at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2218)
                                                                                   at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1917)
                                                                                   at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2218)
                                                                                   at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1917)
                                                                                   at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2218)
                                                                                   at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1917)
                                                                                   at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2218)
                                                                                   at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1917)
                                                                                   at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2218)
                                                                                   at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1917)
                                                                                   at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2218)
                                                                                   at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:1917)
                                                                                   at com.android.internal.policy.impl.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2125)
                                                                                   at com.android.internal.policy.impl.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1572)
                                                                                   at android.app.Activity.dispatchTouchEvent(Activity.java:2535)
                                                                                   at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                                                                                   at com.android.internal.policy.impl.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2073)
                                                                                   at android.view.View.dispatchPointerEvent(View.java:7903)
                                                                                   at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4257)
                                                                                   at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4136)
                                                                                   at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3638)
                                                                                   at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3688)
                                                                                   at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3657)
[... cut ...]
                                                                                   at java.lang.reflect.Method.invoke(Method.java:515)
                                                                                   at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:864)
                                                                                   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:680)
                                                                                   at dalvik.system.NativeStart.main(Native Method)

希望您能够帮忙解决，先谢谢啦!

closed 2017-06-07 by CJT2325