CJT2325/CameraView #50: android 7.0 手机  第一次运行的时候闪退  第二次运行就正常

coolyunyun, 2017-06-02:
android 7.0 手机  第一次运行的时候闪退  第二次运行就正常
targetSdkVersion 25的情况下
前置摄像头点了没反映   
测试版本  1.0.9

E/AndroidRuntime: FATAL EXCEPTION: Thread-19
                                                                java.lang.RuntimeException: Fail to connect to camera service
                                                                    at android.hardware.Camera.<init>(Camera.java:571)
                                                                    at android.hardware.Camera.open(Camera.java:415)
                                                                    at com.cjt2325.cameralibrary.CameraInterface.openCamera(CameraInterface.java:253)
                                                                    at com.cjt2325.cameralibrary.CameraInterface.doOpenCamera(CameraInterface.java:247)
                                                                    at com.cjt2325.cameralibrary.JCameraView$6.run(JCameraView.java:572)

CJT2325, 2017-06-02:
你使用的是我的demo的apk还是自己集成的？

coolyunyun, 2017-06-02:
targetSdkVersion 22就没问题
targetSdkVersion 25的情况第一次就会闪退 第二次运行就正常

closed 2017-06-07 by CJT2325

chenjunyubt, 2018-11-11:
使用现有 1.1.9 也同样还存在这个问题呢？  楼主怎么解决？