CJT2325/CameraView #53: 点击拒绝权限崩溃

luyoulong, 2017-06-05:
06-05 17:38:19.075 7096-7096/com.cjt2325.cameraview E/MessageQueue-JNI: Exception in MessageQueue callback: handleReceiveCallback
06-05 17:38:19.079 7096-7096/com.cjt2325.cameraview E/MessageQueue-JNI: java.lang.RuntimeException: getParameters failed (empty parameters)
                                                                            at android.hardware.Camera.native_getParameters(Native Method)
                                                                            at android.hardware.Camera.getParameters(Camera.java:3098)
                                                                            at com.cjt2325.cameralibrary.CameraInterface.handleFocus(CameraInterface.java:537)
                                                                            at com.cjt2325.cameralibrary.JCameraView.setFocusViewWidthAnimation(JCameraView.java:486)
                                                                            at com.cjt2325.cameralibrary.JCameraView.onTouchEvent(JCameraView.java:424)
                                                                            at android.view.View.dispatchTouchEvent(View.java:8493)
                                                                            at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2540)
                                                                            at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2225)
                                                                            at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2546)
                                                                            at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2182)
                                                                            at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2546)
                                                                            at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2182)
                                                                            at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2546)
                                                                            at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2182)
                                                                            at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2546)
                                                                            at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2182)
                                                                            at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2546)
                                                                            at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2182)
                                                                            at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2546)
                                                                            at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2182)
                                                                            at com.android.internal.policy.impl.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2460)
                                                                            at com.android.internal.policy.impl.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1772)
                                                                            at android.app.Activity.dispatchTouchEvent(Activity.java:2777)
                                                                            at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                                                                            at com.android.internal.policy.impl.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2419)
                                                                            at android.view.View.dispatchPointerEvent(View.java:8694)
                                                                            at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4548)
                                                                            at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4401)
                                                                            at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3939)
                                                                            at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3995)
                                                                            at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3961)
                                                                            at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4071)
                                                                            at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3969)
                                                                            at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4128)
                                                                            at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3939)
                                                                            at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3995)
                                                                            at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3961)
                                                                            at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3969)
                                                                            at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3939)
                                                                            at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6363)
                                                                            at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6319)
                                                                            at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6272)
                                                                            at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6524)
                                                                            at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:193)
                                                                            at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                            at android.os.MessageQueue.next(MessageQueue.java:148)
                                                                            at android.os.Looper.loop(Looper.java:151)
                                                                            at android.app.ActivityThread.main(ActivityThread.java:5572)
                                                                            at java.lang.reflect.Method.invoke(Native Method)
                                                                            at java.lang.reflect.Method.invoke(Method.java:372)
                                                                            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:966)
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:761)

CJT2325, 2017-06-05:
好的，有空修复

CJT2325, 2017-06-12:
已经修复了 @luyoulong

closed 2017-06-12 by CJT2325