CJT2325/CameraView #57: 部分机型录制不流畅

dujiong, 2017-06-08:
用小米,华为还可以录制比较流畅,三星A7录制时,画面不流畅,录制完成播放时没什么问题,很奇怪

CJT2325, 2017-06-08:
(⊙o⊙)… 没遇到过，你先试试用单纯的录视频Demo看看会不会发生这样的情况

yovenny, 2017-06-12:
魅蓝x也感觉到卡顿，具体不知道什么原因。。我看下demo会不会先？？

yovenny, 2017-06-12:
我自己测了跟微信的表现差不多，移动的一刻都会微微的卡一下，可能硬件的问题吧。。。

CJT2325, 2017-06-12:
嗯，可能部分机型会~以后尽力完善····

yovenny, 2017-06-12:
@CJT2325 有没有计划版本的新增功能排期丫

CJT2325, 2017-06-12:
基本功能都差不多了····就不打算新增功能了，后期应该只是维护和完善把~~话说你手机录制的视频的大小大概有多大？ @yovenny

yovenny, 2017-06-12:
默认1.6M ；funy 300k，清晰度差；， 现在用MEDIA_QUALITY_LOW，应该1M差不多吧，没测。但我看我公司以前找的代码，只有几百k，清晰度还可以。具体的代码区别没仔细看，应该也是改了这个videoBitRate
```
mMediaRecorder.setCamera(mCamera);

        // Step 2: Set sources
        try {
            mMediaRecorder.setAudioSource(MediaRecorder.AudioSource.DEFAULT);
            mMediaRecorder.setVideoSource(MediaRecorder.VideoSource.DEFAULT);
        } catch (Exception e) {
            mMediaRecorder.setAudioSource(MediaRecorder.AudioSource.DEFAULT);
            mMediaRecorder.setVideoSource(MediaRecorder.VideoSource.DEFAULT);
            Ln.e("init recorder error:" + e.getMessage());
        }
        try {
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.FROYO) {
                //mMediaRecorder.setVideoSize(640, 480);
                CamcorderProfile lowProfile = CamcorderProfile.get(CamcorderProfile.QUALITY_LOW);
                CamcorderProfile hightProfile = CamcorderProfile.get(CamcorderProfile.QUALITY_HIGH);
                if (lowProfile != null && hightProfile != null) {
                    Log.d("TT", "lowProfile.videoFrameRate:" + lowProfile.videoFrameRate + ", hightProfile.videoFrameRate:" + hightProfile.videoFrameRate);
                    int audioBitRate = lowProfile.audioBitRate > 128000 ? 128000 : lowProfile.audioBitRate;
                    lowProfile.audioBitRate = audioBitRate > hightProfile.audioBitRate ? hightProfile.audioBitRate : audioBitRate;
                    lowProfile.audioSampleRate = 48000 > hightProfile.audioSampleRate ? hightProfile.audioSampleRate : 48000;
//					lowProfile.duration = 20 > hightProfile.duration ? hightProfile.duration : 20;
                    //lowProfile.videoFrameRate = 20 > hightProfile.videoFrameRate ? hightProfile.videoFrameRate : 20;
                    lowProfile.duration = hightProfile.duration;
                    //lowProfile.videoFrameRate = hightProfile.videoFrameRate;
                    lowProfile.videoBitRate = 1500000 > hightProfile.videoBitRate ? hightProfile.videoBitRate : 1500000;

                    if (mDisplayOrientation % 180 == 0) {
                        //lowProfile.videoFrameWidth = mWidth;
                        //lowProfile.videoFrameHeight = mHeight;
                    } else {
                        //lowProfile.videoFrameWidth = mHeight;
                        //lowProfile.videoFrameHeight = mWidth;
                    }
                    lowProfile.videoFrameWidth = mWidth;
                    lowProfile.videoFrameHeight = mHeight;

                    lowProfile.audioCodec = MediaRecorder.AudioEncoder.AAC;
                    lowProfile.videoCodec = MediaRecorder.VideoEncoder.H264; //H264属于MPEG4标准，性能更好
                    //lowProfile.videoCodec = MediaRecorder.VideoEncoder.MPEG_4_SP;
                    //lowProfile.videoFrameRate = 20;
                    lowProfile.videoBitRate = 640 * 1024; //主要是这个影响视频的大小。mMediaRecorder.setVideoEncodingBitRate(5*1024*1024);不过我适当调小了点，要不生成的文件太大了

                    Log.d("TT", "-------------------------------------------------------");
                    Log.d("TT", "lowProfile.audioBitRate:" + lowProfile.audioBitRate);
                    Log.d("TT", "lowProfile.audioChannels:" + lowProfile.audioChannels);
                    Log.d("TT", "lowProfile.audioCodec:" + lowProfile.audioCodec);
                    Log.d("TT", "音频采样率audioSampleRate:" + lowProfile.audioSampleRate);
                    Log.d("TT", "audioBitRate:" + lowProfile.audioBitRate);
                    Log.d("TT", "duration:" + lowProfile.duration);
                    Log.d("TT", "fileFormat:" + lowProfile.fileFormat);
                    Log.d("TT", "quality:" + lowProfile.quality);
                    Log.d("TT", "videoBitRate:" + lowProfile.videoBitRate);
                    Log.d("TT", "videoCodec:" + lowProfile.videoCodec);
                    Log.d("TT", "videoFrameHeight:" + lowProfile.videoFrameHeight);
                    Log.d("TT", "videoFrameWidth:" + lowProfile.videoFrameWidth);
                    Log.d("TT", "视频帧率videoFrameRate:" + lowProfile.videoFrameRate);
                    Log.d("TT", "+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++");

                    mMediaRecorder.setProfile(lowProfile);
                }
            } else { //系统版本低于8时
                setDefaultRecorder();
            }
        } catch (Exception e) {
            setDefaultRecorder();
            Ln.e("init recorder error" + e.getMessage());
        }
[... 28 lines cut ...]
        } catch (IOException e) {
            Ln.e("IOException preparing MediaRecorder: " + e.getMessage());
            releaseMediaRecorder();
            return false;
        } catch (Exception e) {
            Ln.e("Exception preparing MediaRecorder: " + e.getMessage());
            releaseMediaRecorder();
            return false;
        }
        return true;
    }

    private void setDefaultRecorder() {

        mMediaRecorder.setAudioEncoder(MediaRecorder.AudioEncoder.AAC);//mMediaRecorder.setAudioEncoder(MediaRecorder.AudioEncoder.DEFAULT);
        mMediaRecorder.setVideoEncoder(MediaRecorder.VideoEncoder.H264);//mMediaRecorder.setVideoEncoder(MediaRecorder.VideoEncoder.DEFAULT);
        if (mSupportVideoSizes != null && !mSupportVideoSizes.isEmpty()) {
            Collections.sort(mSupportVideoSizes, new SizeComparator());
            Size size = mSupportVideoSizes.get(0);
            mMediaRecorder.setVideoSize(size.width, size.height);
        } else {
            mMediaRecorder.setVideoSize(640, 480); // Its is not on android docs but  it needs to be done. (640x480 = VGA resolution)
        }
    }
```

yovenny, 2017-06-12:
可以折衷一下，尽量减少视频大小，更改影响视频大小的参数到一个临界值，

话说我不知道视频大小的参数有哪些？

huajieqian, 2017-06-14:
魅蓝X的确有录制视频不流畅的问题

CJT2325, 2017-06-14:
播放视频会吗？

huajieqian, 2017-06-14:
也会，就是录制的时候会有瞬间变得很模糊，播的时候也是

CJT2325, 2017-06-14:
噢...那有可能在获取录视频时候的分辨率比较低。我想想如何优化

closed 2017-08-04 by CJT2325