CJT2325/CameraView #59: 初次打开报这个错enable shutter sound failed，第二次以后便不会

yovenny, 2017-06-12:
```
d[Thread-899,5,main]
                                                                 UncaughtException detected: java.lang.RuntimeException: enable shutter sound failed
                                                                     at android.hardware.Camera._enableShutterSound(Native Method)
                                                                     at android.hardware.Camera.enableShutterSound(Camera.java:2052)
                                                                     at com.cjt2325.cameralibrary.CameraInterface.openCamera(CameraInterface.java:280)
                                                                     at com.cjt2325.cameralibrary.CameraInterface.doOpenCamera(CameraInterface.java:265)
                                                                     at com.cjt2325.cameralibrary.JCameraView$5.run(JCameraView.java:422)
06-12 09:50:38.912 25773-29244/com.yovenny.yuyue E/AndroidRuntime: FATAL EXCEPTION: Thread-899
                                                                   Process: com.yovenny.yuyue, PID: 25773
                                                                   java.lang.RuntimeException: enable shutter sound failed
                                                                       at android.hardware.Camera._enableShutterSound(Native Method)
                                                                       at android.hardware.Camera.enableShutterSound(Camera.java:2052)
                                                                       at com.cjt2325.cameralibrary.CameraInterface.openCamera(CameraInterface.java:280)
                                                                       at com.cjt2325.cameralibrary.CameraInterface.doOpenCamera(CameraInterface.java:265)
                                                                       at com.cjt2325.cameralibrary.JCameraView$5.run(JCameraView.java:422)
```

CJT2325, 2017-06-12:
嗯，我之前已经修复了这个问题，代码更新了，但是Demo还么更新，我刚刚上传了最新的Demo了，你重新下载试试看

CJT2325, 2017-06-12:
JCenter版本可以也有点Bug，等修复完BUG之后，再更新一个版本 @yovenny

yovenny, 2017-06-12:
我是昨天拉下来加入module的，我现在再试试哈，谢谢呀

CJT2325, 2017-06-12:
我重新添加了捕获异常，现在应该不会崩溃了

yovenny, 2017-06-12:
那边捕获？，没看到相关代码的，这个是什么原因丫

yovenny, 2017-06-12:
哦哦，看到了。。

closed 2017-06-12 by yovenny