CJT2325/CameraView #60: 更改录制框的大小？有没有计划加入这个高度自定义？

yovenny, 2017-06-12:
[image]

photoParam.setMargins(0, 0, 0, mCaptureLayout.getCaptureHeight());
 videoViewParam.setMargins(0, 0, 0, mCaptureLayout.getCaptureHeight());
我改了这两个，ui是生效了，但感觉录制的大小还是一样，要改哪些录制参数呀？告诉我呗，

有没有计划加入这个高度自定义？

CJT2325, 2017-06-12:
那肯定，你要去cameraInterface的startRecorder里面设置分辨率

yovenny, 2017-06-12:
改完报错，尴尬 
previewSize.height -=mCatureHeight*pictureSize.height/mScreenheigth;
这个分辨率怎么改？@CJT2325 大大

CJT2325, 2017-06-12:
这个分辨率不能随意设置的，要根据你手机支持的分辨率来设置

yovenny, 2017-06-12:
具体代码我应该怎么写，就录制视频后获取某一帧作为封面，它的的高度怎么还是为1280，换算后高度肯定不是1280，那这个值具体代码应该减去这个值怎么算？@CJT2325

CJT2325, 2017-06-12:
获取你修改后的长宽比去获取最接近的分辨率，我里面有个类专门获取分辨率的

yovenny, 2017-06-12:
好

yovenny, 2017-06-12:
但之前我debug分辨率列表只有size=10,明天继续看看 thanks

yovenny, 2017-06-13:
谢谢作者

```java
    @Override
    protected void onMeasure(int widthMeasureSpec, int heightMeasureSpec) {
        super.onMeasure(widthMeasureSpec, heightMeasureSpec);
        float widthSize = MeasureSpec.getSize(widthMeasureSpec);
        float heightSize = MeasureSpec.getSize(heightMeasureSpec);
        heightSize-=mCaptureLayout.getCaptureHeight();
        screenProp = heightSize / widthSize;
    }
```

closed 2017-06-13 by yovenny