CJT2325/CameraView #61: AutoFocus problem

vellrya, 2017-06-14:
Hello! I tried your library on OnPlus 3. The maximum resolution of the preview was only 1280 * 720, although the standard camera gives out Full HD. In addition, autofocus does not work. You can only focus by touch.
Is there a solution?

CJT2325, 2017-06-14:
I have fixed the BUG，but haven't update to Jcenter

vellrya, 2017-06-14:
I do not use it. I downloaded the latest version here, on github.

CJT2325, 2017-06-14:
can you show me the log？

vellrya, 2017-06-14:
@CJT2325, yes, of course)
`06-14 15:52:29.162 25777-25777/ru.vellrya.bigbrain I/CJT: surfaceCreated
06-14 15:52:29.490 25777-25834/ru.vellrya.bigbrain I/JCameraView: FocusMode supported auto
06-14 15:52:29.490 25777-25834/ru.vellrya.bigbrain I/JCameraView: Formats supported 256
06-14 15:52:29.665 25777-25834/ru.vellrya.bigbrain I/CJT: === Start Preview ===
06-14 15:52:30.726 25777-25777/ru.vellrya.bigbrain E/ExtMediaPlayer-JNI: env->IsInstanceOf fails
06-14 15:52:30.726 25777-25777/ru.vellrya.bigbrain E/MediaPlayer-JNI: JNIMediaPlayerFactory: bIsQCMediaPlayerPresent 0
06-14 15:52:30.726 25777-25777/ru.vellrya.bigbrain E/ExtMediaPlayer-JNI: env->IsInstanceOf fails
06-14 15:52:30.726 25777-25777/ru.vellrya.bigbrain E/MediaPlayer-JNI: JNIMediaPlayerFactory: bIsQCMediaPlayerPresent 0
06-14 15:52:30.764 25777-25777/ru.vellrya.bigbrain D/MediaPlayer: setSubtitleAnchor in MediaPlayer
06-14 15:52:31.281 25777-25792/ru.vellrya.bigbrain E/Camera-JNI: copyAndPost: off=0, size=875863 msg=100
06-14 15:52:31.547 25777-25777/ru.vellrya.bigbrain D/Camera: app passed NULL surface
06-14 15:52:31.774 25777-25777/ru.vellrya.bigbrain I/CJT: === Stop Camera ===
06-14 15:52:32.108 25777-25777/ru.vellrya.bigbrain I/Choreographer: Skipped 49 frames!  The application may be doing too much work on its main thread.
06-14 15:52:32.131 25777-25806/ru.vellrya.bigbrain D/AppTracker: App Event: stop
06-14 15:52:32.255 25777-25777/ru.vellrya.bigbrain D/CropIwaLog: LoadBitmapCommand for file:///storage/emulated/0/Android/data/ru.vellrya.bigbrain/files/data/captured_photo.jpg delayed, wrong dimensions {width=0, height=0}
06-14 15:52:32.261 25777-25824/ru.vellrya.bigbrain D/AppTracker: App Event: start
06-14 15:52:32.295 25777-25777/ru.vellrya.bigbrain D/CropIwaLog: load bitmap request for {file:///storage/emulated/0/Android/data/ru.vellrya.bigbrain/files/data/captured_photo.jpg}
06-14 15:52:32.326 25777-25777/ru.vellrya.bigbrain W/art: Before Android 4.1, method int android.support.v7.widget.ListViewCompat.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
06-14 15:52:32.366 25777-25827/ru.vellrya.bigbrain D/CropIwaLog: loaded image with dimensions {width=1080, height=1920}
06-14 15:52:32.394 25777-25777/ru.vellrya.bigbrain D/CropIwaLog: {file:///storage/emulated/0/Android/data/ru.vellrya.bigbrain/files/data/captured_photo.jpg} loading completed, listener got the result
06-14 15:52:32.445 25777-25777/ru.vellrya.bigbrain I/CJT: surfaceDestroyed`

CJT2325, 2017-06-15:
i got what do you mean，my lib's autofocus does work when recording video,  focus by touch does work when capture，and resolution is the resolution of a length to width ratio that is closest to the screen of a cell phone.

vellrya, 2017-06-15:
Thank you very much! I just turned on the autofocus for pictures by adding this code:
`if (CameraParamUtil.getInstance().isSupportedFocusMode(
                        mParams.getSupportedFocusModes(),
                        Camera.Parameters.FOCUS_MODE_CONTINUOUS_PICTURE)) {
                    mParams.setFocusMode(Camera.Parameters.FOCUS_MODE_CONTINUOUS_PICTURE);
                } else if (CameraParamUtil.getInstance().isSupportedFocusMode(
                        mParams.getSupportedFocusModes(),
                        Camera.Parameters.FOCUS_MODE_AUTO)) {
                    mParams.setFocusMode(Camera.Parameters.FOCUS_MODE_AUTO);
                }`
The problem is solved!

closed 2017-06-15 by vellrya