CJT2325/CameraView #75: Capture button chinese font

ibrahimuruker, 2017-07-09:
How to edit,chinese text on the take button?

nemesistrk, 2017-07-10:
Hi, we need this library with language support. You have some Chinese words in your main code which we cant edit that texts. 

( Ex;  https://github.com/CJT2325/CameraView/blob/master/camera/src/main/java/com/cjt2325/cameralibrary/CaptureLayout.java  line 275 )

Can you move that words in a XML file, so if anyone want to change it for their language, it should be very useful. 

Thanks.

CJT2325, 2017-07-11:
I provided a function 'setTip(String)' to set the text. @ibrahimuruker

ibrahimuruker, 2017-07-11:
How to use this function @CJT2325 
I get an error when I use: jCameraView.setTip("Bla Bla Bla");

CJT2325, 2017-07-11:
HAHA~Are you using the 1.1.2 version?you can try the 1.1.3 version bacuse I forgot to update Readme~ @ibrahimuruker

ibrahimuruker, 2017-07-11:
O very good. But:
Yes, this function works in version 1.1.3.
However, JCameraLisenter () started to give an error in 1.1.3 version

ibrahimuruker, 2017-07-11:
I found the cause of the error. You changed the recordSuccess () method,
I made the necessary changes to the exchange
thank you so much...

by the way
Do you think to add an flash button?

CJT2325, 2017-07-11:
Yes,I will add this module when I have free time,
Thank you for your support,because I am Chinese, maybe my English  difficult to understand, ha ha

ibrahimuruker, 2017-07-11:
I thank you very much
You did a very nice job.

closed 2017-08-03 by CJT2325