CJT2325/CameraView #78: home键再回来闪退

312353457, 2017-07-13:
-13 13:10:11.182 5594-5665/? E/CameraService: disconnect: Disconnect called on already disconnected client for device 0
07-13 13:10:11.183 28127-28127/com.cjt2325.cameraview W/System.err: java.lang.RuntimeException: getParameters failed (empty parameters)
07-13 13:10:11.195 28127-28127/com.cjt2325.cameraview W/System.err:     at android.hardware.Camera.native_getParameters(Native Method)
07-13 13:10:11.195 28127-28127/com.cjt2325.cameraview W/System.err:     at android.hardware.Camera.getParameters(Camera.java:2272)
07-13 13:10:11.195 28127-28127/com.cjt2325.cameraview W/System.err:     at com.cjt2325.cameralibrary.CameraInterface.doStartPreview(CameraInterface.java:344)
07-13 13:10:11.195 28127-28127/com.cjt2325.cameraview W/System.err:     at com.cjt2325.cameralibrary.JCameraView.cameraHasOpened(JCameraView.java:412)
07-13 13:10:11.195 28127-28127/com.cjt2325.cameraview W/System.err:     at com.cjt2325.cameralibrary.CameraInterface.doOpenCamera(CameraInterface.java:289)
07-13 13:10:11.195 28127-28127/com.cjt2325.cameraview W/System.err:     at com.cjt2325.cameralibrary.JCameraView.handlerPictureOrVideo(JCameraView.java:627)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at com.cjt2325.cameralibrary.JCameraView.access$1700(JCameraView.java:45)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at com.cjt2325.cameralibrary.JCameraView$3.confirm(JCameraView.java:386)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at com.cjt2325.cameralibrary.CaptureLayout$4.onClick(CaptureLayout.java:241)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at android.view.View.performClick(View.java:5264)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at android.view.View$PerformClick.run(View.java:21297)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at android.os.Handler.handleCallback(Handler.java:743)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at android.os.Looper.loop(Looper.java:150)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5621)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:794)
07-13 13:10:11.196 28127-28127/com.cjt2325.cameraview W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:684)

CJT2325, 2017-07-13:
是什么状态下按home键的

312353457, 2017-07-18:
两种情况都闪退，1.进入录视频界面，然后Home，再回来录就闪退了。2.录完视频，按home,然后再回来，点击返回或确定，闪退！ 测试你的demo也是一样的情况！机型是华为荣耀8

closed 2017-09-13 by CJT2325