CJT2325/CameraView #89: 必现bug

RelocationTime, 2017-08-25:
打开CameraActivity , 然后按onback , 又打开camearActivity拍照，又onback， 过程要快， 二到三次必carsh

RelocationTime, 2017-08-25:
time: FATAL EXCEPTION: Thread-31943
                                                            Process: im.chaoxin, PID: 22165
                                                            java.lang.RuntimeException: Method called after release()
                                                                at android.hardware.Camera.native_getParameters(Native Method)
                                                                at android.hardware.Camera.getParameters(Camera.java:3231)
                                                                at im.chaoxin.chat.takemedia.CameraInterface.handleFocus(CameraInterface.java:704)
                                                                at im.chaoxin.chat.takemedia.JCameraView.cameraHasOpened(JCameraView.java:432)
                                                                at im.chaoxin.chat.takemedia.CameraInterface.doOpenCamera(CameraInterface.java:281)
                                                                at im.chaoxin.chat.takemedia.JCameraView$8.run(JCameraView.java:662)

CJT2325, 2017-08-25:
em~我研究研究

RelocationTime, 2017-08-25:
我试过在 activity ondestory销毁camera,也为不行， 在打开camera调用单例去销毁camear也不行，不知道问题出现在哪里

CJT2325, 2017-08-25:
嗯，因为后面一点一点功能加上去，代码有点冗余，最近在更新使用Camera2，到时候一起把这个Camera1的给重构一下

RelocationTime, 2017-08-25:
那楼主，这个bug 可以fix 吗， 现在

RelocationTime, 2017-08-25:
感谢 《 - 》

RelocationTime, 2017-08-25:
fix bug 解决 了

  protected void onDestroy() {
        super.onDestroy();
        if (jCameraView!=null) {
            jCameraView.stopRecording();
        }
    }
就可以了，谢谢

RelocationTime, 2017-08-25:
好像还不行。。。。

CJT2325, 2017-08-25:
哈哈  我有空看看把  我已经下班了  周末回去看

RelocationTime, 2017-08-25:
那么 早啊

RelocationTime, 2017-08-25:
不是996

CJT2325, 2017-08-25:
我965  哈哈

closed 2017-09-13 by CJT2325