CJT2325/CameraView #96: 录制时间1s限制问题

Leekalun, 2017-09-15:
请问一下大神  我想录制时间小于一秒时候 变成拍照效果，怎么实现呢

CJT2325, 2017-09-15:
你试试在JCameraView 里面 的recordShort这个回调里面，尝试调用一下 match.capture

Leekalun, 2017-09-15:
其他手机都可以了，唯独三星galaxy note3 不行，而且长按录制之后，按钮就不能点击了

CJT2325, 2017-09-15:
不行是指？

Leekalun, 2017-09-15:
这个型号的手机， 就算我不更改代码，长按录制视频的时候，都会卡顿2-3s, 现在我改成1秒内match,capture之后， 其他手机都可以。三星的就是恢复预览状态，button 也不能点击不能长按

CJT2325, 2017-09-15:
.......那么萌  button不能点击 可能是里面state设置为了STATE_BAN了   你给我看看你怎么改的

Leekalun, 2017-09-15:
[image]
您看，我就简单的改了一下， 主要是 只有Galaxy note3 的手机不行

Leekalun, 2017-09-15:
[image]
您请看， 我就这样简单的加个方法， 唯独它不行

CJT2325, 2017-09-15:
你应该加在run的 这里会出问题的

Leekalun, 2017-09-15:
放在里面，三星还是不行, 其他手机都没问题，

Leekalun, 2017-09-15:
我下载了demo, 现在的bug 就是 长按录制视频， 然后取消，之后按钮就不可点击了（galaxy note3）不知道是不是手机的问题，无解

CJT2325, 2017-09-15:
就这台是吗？？？有报错log什么的吗

Leekalun, 2017-09-15:
[image]
对， 就一台手机是这种情况

CJT2325, 2017-09-15:
什么系统的

Leekalun, 2017-09-15:
安卓4.4.2的

closed 2018-07-26 by CJT2325