CJT2325/CameraView #105: Error quit() not found

Paul75, 2017-10-07:
Hello,

I add o gradle `compile 'cjt.library.wheel:camera:1.1.9'` but the quit() not found in JCameraListener

CJT2325, 2017-10-07:
In version 1.1.9, I replace JCameraListener.quit () with ClickListener.onClick ()
You can see the README
```java
jCameraView.setJCameraLisenter(new JCameraLisenter() {
    @Override
    public void captureSuccess(Bitmap bitmap) {
        //获取图片bitmap
        Log.i("JCameraView", "bitmap = " + bitmap.getWidth());
    }
    @Override
    public void recordSuccess(String url) {
        //获取视频路径
        Log.i("CJT", "url = " + url);
     }
    //@Override
    //public void quit() {
    //    (1.1.9+后用左边按钮的点击事件替换)
    //}
});
jCameraView.setLeftClickListener(new ClickListener() {
    @Override
    public void onClick() {
        CameraActivity.this.finish();
    }
});
jCameraView.setRightClickListener(new ClickListener() {
    @Override
        Toast.makeText(CameraActivity.this,"Right",Toast.LENGTH_SHORT).show();
    }
});
```

Paul75, 2017-10-07:
yes but JCameraLisenter not found ....

CJT2325, 2017-10-07:
JCameraLisenter modify to JCameraListener ...

Paul75, 2017-10-07:
i do this but quit not found ....

Paul75, 2017-10-07:
if i go to the source it have :

`package com.cjt2325.cameralibrary.listener;

import android.graphics.Bitmap;

/**
 * =====================================
 * 作    者: 陈嘉桐
 * 版    本：1.1.4
 * 创建日期：2017/4/26
 * 描    述：
 * =====================================
 */
public interface JCameraListener {

    void captureSuccess(Bitmap bitmap);

    void recordSuccess(String url, Bitmap firstFrame);

}
`

CJT2325, 2017-10-07:
You just delete this quit method
Are you running applicationcamera model?

CJT2325, 2017-10-07:
Because I've just updated a new version and replaced the quit with ClickListener.onClick ()

Paul75, 2017-10-07:
I juste import the gradle and in my activity :

`jCameraView = findViewById(R.id.jcameraview);

        jCameraView.setSaveVideoPath(Environment.getExternalStorageDirectory().getPath() + File.separator + "GOMPOST_Photos");

        jCameraView.setJCameraLisenter(new JCameraListener() {
            @Override
            public void captureSuccess(Bitmap bitmap) {
                Log.i("JCameraView", "bitmap = " + bitmap.getWidth());
            }

            @Override
            public void recordSuccess(String url, Bitmap firstFrame) {
                Log.i("CJT", "url = " + url);
            }

            @Override
            public void quit() {
                JCameraViewActivity.this.finish();
            }
        });

        getPermissions();`

and permision method and the rest

Paul75, 2017-10-07:
yes I understand. sorry

CJT2325, 2017-10-07:
Maybe my English is poor, ha ha, you'd better according to applicationcamera model to write

Paul75, 2017-10-07:
but why when I implement this, all of my icon in the layout (icon camera)
change ?

CJT2325, 2017-10-07:
Sorry, I can't understand that...

Paul75, 2017-10-07:
in my activity first, i have my icons of camera.
but when i implement your library it change all icon mine.

Paul75, 2017-10-07:
because in drawable I have mine ic_camera name ?

CJT2325, 2017-10-07:
Mabey，my drawable icon name is ic_cmaera too

CJT2325, 2017-10-07:
you can try to change your drawable name

Paul75, 2017-10-07:
also Have a question :

can I change icon left ? why ? the the icon <- ?
when I can change message on the top of button take photo ?
Thanks

CJT2325, 2017-10-07:
```java
<com.cjt2325.cameralibrary.JCameraView
    android:id="@+id/jcameraview"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    app:duration_max="10000"
    app:iconMargin="20dp"
    app:iconSize="30dp"
    app:iconLeft="@drawable/ic_camera_enhance_black_24dp"
    app:iconRight="@drawable/ic_camera_enhance_black_24dp"
    app:iconSrc="@drawable/ic_camera_enhance_black_24dp"/>
```
setTip(String tip) to change message on the top of button

Paul75, 2017-10-08:
Helo,

sorry but today I relanch it and now when take photo and validate, it block the camera and nothing do (not save to file) and in log appear not entry to ::
`@Override
            public void captureSuccess(Bitmap bitmap) {
                Log.i("JCameraView", "bitmap = " + bitmap.getWidth());
            }

            @Override
            public void recordSuccess(String url, Bitmap firstFrame) {
                Log.i("JCameraView", "url = " + url);
            }`

CJT2325, 2017-10-08:
Did you click the green arrow button on the right?

Paul75, 2017-10-08:
Sorry I'm wrong. It works fine I delete the actions. I do like your exemple cameraapplication. And it works.

But why i must have in captureSuccess 
`Log.i("JCameraView", "bitmap = " + bitmap.getWidth());
                String path = FileUtil.saveBitmap("GOMPOST_Photos", bitmap);

                Intent returnIntent = new Intent();
                returnIntent.putExtra("result",path);
                returnIntent.putExtra("pneu",pneuChoice);
                setResult(Activity.RESULT_OK,returnIntent);
                finish();`

and in recordSuccess :::
`String path = FileUtil.saveBitmap("GOMPOST_Photos", firstFrame);
                Log.i("CJT", "url = " + url + ", Bitmap = " + path);
                Intent returnIntent = new Intent();
                returnIntent.putExtra("result",path);
                returnIntent.putExtra("pneu",pneuChoice);
                setResult(Activity.RESULT_OK,returnIntent);
                finish();`

because if I return only path ?

CJT2325, 2017-10-08:
Sorry, I don't know what you mean...

Paul75, 2017-10-08:
Why I return in your sample to intent ?

Paul75, 2017-10-08:
It works and not a problem but now I add 👍 
jCameraView.setMediaQuality(JCameraView.MEDIA_QUALITY_HIGH);

but when I see fils it done only about 400 ko ? normal ?

It see not a good quality of image ?

CJT2325, 2017-10-08:
jCameraView.setMediaQuality(JCameraView.MEDIA_QUALITY_HIGH)  just set the quality of video

CJT2325, 2017-10-08:
There are a lot of parameters that affect the quality of the photos, but I haven't dealt with them yet

CJT2325, 2017-10-08:
Does anyone in your country use WeChat?

Paul75, 2017-10-08:
because I do a photo and see that it was blurring ?

Paul75, 2017-10-08:
Yes I have wechat

CJT2325, 2017-10-08:
ha~ Can I add you as a friend?

Paul75, 2017-10-08:
of course
my id wechat is : PaulB75

CJT2325, 2017-10-08:
i will handle with the quality of photo in next version

closed 2017-10-08 by Paul75