CJT2325/CameraView #106: Error in captureSuccess : E/JavaBinder: !!! FAILED BINDER TRANSACTION !!!  (parcel size = 3686704)

Paul75, 2017-10-07:
Hello,

I do this :
`public void captureSuccess(Bitmap bitmap) {
                Log.i("JCameraView", "bitmap = " + bitmap.getWidth());
                Intent returnIntent = new Intent();
                returnIntent.putExtra("result",bitmap);
                returnIntent.putExtra("pneu",pneuChoice);
                setResult(Activity.RESULT_OK,returnIntent);
                JCameraViewActivity.this.finish();
            }`
 
to return in my activity result the bitmap. But I have in a  console :
E/JavaBinder: !!! FAILED BINDER TRANSACTION !!!  (parcel size = 3686704)

Why ?

Paul75, 2017-10-08:
you don't know ?

CJT2325, 2017-10-08:
Intent cannot transfer large data ...

Paul75, 2017-10-08:
but with the demo of cameraview of google it not pose problem to transfert the image ?

CJT2325, 2017-10-08:
you just need to save the Bitmap as file and transfer the url of the file by Intent

CJT2325, 2017-10-08:
url of the file  or   url of one ? Which is better?  haha~

Paul75, 2017-10-08:
Thanks for help

closed 2017-10-08 by Paul75