CJT2325/CameraView #113: 拍摄时间不准确

shaonialife, 2017-10-25:
设置15秒的拍摄时间，在我的荣耀7和 vivo 手机上时间总是短，不到15秒，即使加了 500毫秒用来四舍五入也还是有问题，在小米手机上没什么问题，后来将 CaptureButton 里的 CountDownTimer 换成了 ValueAnimator 在加上500毫秒的四舍五入就好了，代码如下
```java
   private ValueAnimator timer;                     //计时器

    /**
     * 录制视频计时器
     */
    private ValueAnimator createTimerAnimator(final long duration){
        ValueAnimator animator = new ValueAnimator();
        animator.setDuration(duration).setIntValues(0, 360);
        animator.setInterpolator(new LinearInterpolator());
        animator.addUpdateListener(new ValueAnimator.AnimatorUpdateListener() {
            @Override
            public void onAnimationUpdate(ValueAnimator animation) {
                Object value = animation.getAnimatedValue();
                if (value != null && value instanceof Integer){
                    updateProgress((Integer) value, animation.getCurrentPlayTime());
                }
            }
        });
        animator.addListener(new AnimatorListenerAdapter() {
            @Override
            public void onAnimationEnd(Animator animation) {
                updateProgress(360, duration);
                recordEnd();
            }
        });
        return animator;
    }

    /**
     * 更新进度条
     */
    private void updateProgress(int degree, long time) {
        recordedTime = (int) time;
        progress = degree;
        invalidate();
    }
```
创建定时器如下
```java
    timer = createTimerAnimator(duration); //录制定时器
```

CJT2325, 2017-10-25:
我2.4版本把ValueAnimation 改为 CountDownTimer  因为如果手机关了动画过渡效果的话，录制会出现问题

shaonialife, 2017-10-25:
果然，我还是改回去吧，ValueAnimator 里面有一个 缩放因子 sDurationScale，由系统控制，控制了动画的实际时长，修改 设置-开发者选项-动画程序时长调整，就会改变 sDurationScale

shaonialife, 2017-10-27:
真正的原因找到了，不是 CountDownTimer 的原因，而是在某些手机上 MediaRecorder 开始录制会有延时，而且没有回调方法告知程序员实际什么时候开始，很早就有这个问题了，解决的方法是只能换用其它的视频录制方法，坑爹的谷歌，坑爹的OEM

closed 2017-11-03 by CJT2325