CJT2325/CameraView #128: 画质开最高为什么还是这么渣，这个画质跟微信的画质不是一个级别的啊

yuanmenghao, 2017-11-27:


dom4j1, 2017-12-15:
请问这个问题 您解决了么

dom4j1, 2017-12-15:
@yuanmenghao   请问您最后怎么解决的

yuanmenghao, 2017-12-15:
没有，暂时放下这个问题了，画质渣的不行，你可以换个其他框架看看。

yuanmenghao, 2017-12-15:
有好的框架了可以推荐给我一下

dom4j1, 2017-12-15:
@yuanmenghao   这个框架提供了设置视频质量的方法 设置成最高  感觉也没什么变化

dom4j1, 2017-12-15:
@yuanmenghao   我看下其他的框架吧

yuanmenghao, 2017-12-15:
是的，我没时间看源码，推断应该是压缩视频的步骤有问题 或者设置码率的代码没生效。时间充裕的话你可以把源码下载出来自己改一下

dom4j1, 2017-12-15:
@yuanmenghao  我目前正在修改他的源码 如果解决了 第一时间通知你

yuanmenghao, 2017-12-15:
好的，谢谢

dom4j1, 2017-12-15:
10秒的小视频 最大的体积 你能接受的是多少M

dom4j1, 2017-12-18:
通过修改源码 我已解决 视频录制不清晰问题  不过录制出来的视频会变大 如果您需要  可以随时联系我

yuanmenghao, 2017-12-18:
好的。谢谢。

wuxiaoqiang625, 2018-01-05:
想问下 ，什么位置修改视频录制不清晰问题，也想改下 @dom4j1

dom4j1, 2018-01-15:
@wuxiaoqiang625  sorry  如果您还需要修改 可以加我qq 1464529456 我截图给您 文字无法具体描述

zhxcryptic, 2018-01-30:
jCameraView.setMediaQuality(JCameraView.MEDIA_QUALITY_MIDDLE);
画质设置high 比特率为200000 如果不行可以继续设置 数据越大画质越好
jCameraView.setMediaQuality(400000); 大小约为4M 
jCameraView.setMediaQuality(800000); 大小约为8M

dom4j1, 2018-01-30:
@zhxcryptic  这个确实是影响视频画质的原因之一 但是这个框架录制出来的视频不清晰除了修改这个比特率外 还需要修改别的地方 单独修改比特率 即使改到最大也还是不清晰

dom4j1, 2018-01-30:
也就是说这个框架录制出的视频不清晰 真正的原因并不是比特率大小的问题 而是视频分辨率的问题

irly01, 2018-01-30:
原因是软件获取的系统设置分辨率过小。

yuanmenghao, 2018-04-01:
@dom4j1 兄弟给我说下 你修改的是哪段代码，我现在需要用这个了，感谢。。

andyxialm, 2018-04-25:
1. 调整视频录制输出宽高 CameraInterface.java
`videoSize = CameraParamUtil.getInstance().getPreviewSize(mParams.getSupportedPreviewSizes(), 600, screenProp);`
2. `JCameraView.setMediaQuality(int quality)`

closed 2018-07-26 by CJT2325