CJT2325/CameraView #132: 第二次进入拍照界面崩溃

JamesGZM, 2017-12-07:
机型是红米note3高配版
android 6.0.1

JamesGZM, 2017-12-07:
java.lang.RuntimeException: setParameters failed
at android.hardware.Camera.native_setParameters(Native Method)
at android.hardware.Camera.setParameters(Camera.java:1972)
at com.cjt2325.cameralibrary.CameraInterface.doStartPreview(CameraInterface.java:391)
at com.cjt2325.cameralibrary.JCameraView.cameraHasOpened(JCameraView.java:279)
at com.cjt2325.cameralibrary.CameraInterface.doOpenCamera(CameraInterface.java:302)
at com.cjt2325.cameralibrary.JCameraView$7.run(JCameraView.java:307)

CJT2325, 2017-12-07:
运行demo的吗

JamesGZM, 2017-12-07:
@CJT2325 直接导入按照demo写的代码

CJT2325, 2017-12-07:
你运行那个model？

CJT2325, 2017-12-07:
运行那个cameraapplication这个

JamesGZM, 2017-12-07:
@CJT2325 摄像头是正常打开的 即时报了崩溃信息也打开了。。
还有一种情况会报 Thread ID: 31055 Thread Name:　pool-1-thread-1 Time: 2017-12-07 19:45:00:24 FromClass: CrashHandler > ↓↓↓↓exception↓↓↓↓
java.lang.RuntimeException: getParameters failed (empty parameters)
at android.hardware.Camera.native_getParameters(Native Method)
at android.hardware.Camera.getParameters(Camera.java:1984)
at com.cjt2325.cameralibrary.CameraInterface.doStartPreview(CameraInterface.java:368)
at com.cjt2325.cameralibrary.JCameraView.cameraHasOpened(JCameraView.java:279)
at com.cjt2325.cameralibrary.CameraInterface.doOpenCamera(CameraInterface.java:302)
at com.cjt2325.cameralibrary.JCameraView$7.run(JCameraView.java:307)

JamesGZM, 2017-12-07:
@CJT2325 编译中.

JamesGZM, 2017-12-07:
@CJT2325  cameraapplication直接报
java.lang.RuntimeException: startPreview failed
at android.hardware.Camera.startPreview(Native Method)
at com.cjt2325.cameralibrary.CameraInterface.doStartPreview(CameraInterface.java:396)
at com.cjt2325.cameralibrary.JCameraView.cameraHasOpened(JCameraView.java:279)
at com.cjt2325.cameralibrary.CameraInterface.doOpenCamera(CameraInterface.java:302)
at com.cjt2325.cameralibrary.JCameraView$7.run(JCameraView.java:307)

JamesGZM, 2017-12-11:
@CJT2325  
[image]
我fork了你的代码 进行修改 成功运行。。。。 不会报那些错了

closed 2019-01-16 by CJT2325