CJT2325/CameraView #139: OutOfMemory

ZhuDaiLei, 2018-01-05:
java.lang.OutOfMemoryError: Failed to allocate a 3686412 byte allocation with 2614424 free bytes and
多次拍照会出现OOM，请求的内存过大，申请失败，请修复一下哇，^_^

ZhuDaiLei, 2018-01-05:
java.lang.OutOfMemoryError: Failed to allocate a 3686412 byte allocation with 1863840 free bytes and 1820KB until OOM
                                                                        at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                        at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                        at android.graphics.Bitmap.createBitmap(Bitmap.java:977)
                                                                        at android.graphics.Bitmap.createBitmap(Bitmap.java:948)
                                                                        at android.graphics.Bitmap.createBitmap(Bitmap.java:879)
                                                                        at com.cjt2325.cameralibrary.CameraInterface$2.onPictureTaken(CameraInterface.java:481)
                                                                        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1115)
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                        at android.os.Looper.loop(Looper.java:154)
                                                                        at android.app.ActivityThread.main(ActivityThread.java:6832)
                                                                        at java.lang.reflect.Method.invoke(Native Method)
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1543)
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1431)

ZhuDaiLei, 2018-01-05:
当设置为只能拍照时，提示信息也是轻触拍照长按摄像，修改提示信息的方法可否提供出来呢

ZhuDaiLei, 2018-01-09:
在拍照的时候，还可以点击左边的按键，返回过后程序会崩溃

ZhuDaiLei, 2018-01-09:
是这个问题The surface has been released

lvxiaolongde, 2018-07-06:
oom问题

lvxiaolongde, 2018-07-06:
oom问题解决方法，将cameraparamutil中的getPictureSize()方法修改为
public static Camera.Size getSmallPictureSize(Camera camera){
        if(camera != null){
            List<Camera.Size> sizes = camera.getParameters().getSupportedPictureSizes();
            Camera.Size temp = sizes.get(0);
            for(int i = 1;i < sizes.size();i ++){
                float scale = (float)(sizes.get(i).height) / sizes.get(i).width;
                if(temp.width > sizes.get(i).width && scale < 0.6f && scale > 0.5f)
                    temp = sizes.get(i);
            }
            Log.i(TAG, "MakeSure Picture :w = " + temp.width + " h = " + temp.height);
            return temp;
        }
        return null;
    }
亲测可行！

closed 2019-01-16 by CJT2325