CJT2325/CameraView #153: 切换到前置摄像头时,点击切换闪光灯效果,闪退

wfqliuchao, 2018-01-31:


wfqliuchao, 2018-01-31:
//切换摄像头
        mSwitchCamera.setOnClickListener(new OnClickListener() {
            @Override
            public void onClick(View v) {
                machine.swtich(mVideoView.getHolder(), screenProp);
                Camera.CameraInfo info = new Camera.CameraInfo();
                int count = Camera.getNumberOfCameras();
                for (int i = 0; i < count; i++) {
                    Camera.getCameraInfo(i, info);
                    if (info.facing == Camera.CameraInfo.CAMERA_FACING_FRONT) {
                        mFlashLamp.setVisibility(GONE);
                    } else {
                        mFlashLamp.setVisibility(VISIBLE);
                    }
                }
            }
        });
这样就OK了

Cazaea, 2018-03-28:
你这样虽然解决了ANR的问题，但是，并不能保证在切换摄像头之后，能够正确的显示隐藏闪光灯的ImageView。

closed 2019-01-16 by CJT2325