CJT2325/CameraView #162: 关于放大摄像头问题

Alvin9234, 2018-03-17:
if (!mParams.isZoomSupported() || !mParams.isSmoothZoomSupported()) {
            return;
 }
isZoomSupported返回true，isSmoothZoomSupported平滑缩放返回false，所以放大摄像头在某些手机上不起作用。其实手机支持放大摄像头的，微信的也是可以缩放摄像头，不清楚是不是不是没有这个判断，如果去掉isSmoothZoomSupported这个判断，就可以缩放了，所以加上这个判断的原因是什么呢

Alvin9234, 2018-03-17:
我觉得是不是可以将这个判断改成
if (!mParams.isZoomSupported()) {
return;
}
然后在进行缩放的时候判断一下
if(mParams.isSmoothZoomSupported()){//支持平滑缩放则使用平滑缩放
        mCamera.startSmoothZoom(scaleRate);
 }else{
        mParams.setZoom(scaleRate);
        mCamera.setParameters(mParams);
}

BuleB, 2018-03-21:
亲，按照您的方法，我在
```
if (scaleRate <= mParams.getMaxZoom() && scaleRate >= nowScaleRate && recordScleRate != scaleRate) {
if (mParams.isSmoothZoomSupported()) {
     mCamera.startSmoothZoom(scaleRate);
 else {
     mParams.setZoom(scaleRate);
     mCamera.setParameters(mParams);
      }
      recordScleRate = scaleRate;
}
```
加上了上述的方法，并且修改了
```
 if (!mParams.isZoomSupported()) {
    return;
}
```
然后就抛出异常提示：设置参数失败的异常setParameters failed

然后我还在找解决的方法 @Alvin9234

BuleB, 2018-03-21:
忘记贴出我的手机型号了，小米4，移动版，OS:Android6.0.1,MIUI:9.5稳定版

Alvin9234, 2018-03-21:
setParameters failed的原因网上大多说的是设置预览分辨率问题，你看一下这http://blog.csdn.net/never_said_no/article/details/78683629，我上面的描述只是对摄像头的缩放判断觉得可以改，这篇博客里面有提到这个setPictureSize(previewSize.width, previewSize.height)，我看了一下代码，这个demo里是有这种设置预览的代码的

closed 2019-01-16 by CJT2325