CJT2325/CameraView #199: 报错,两个界面一个拍照一个拍视频,会报错

kevin321happy, 2018-08-27:
java.lang.RuntimeException: getParameters failed (empty parameters)
                                                                        at android.hardware.Camera.native_getParameters(Native Method)
                                                                        at android.hardware.Camera.getParameters(Camera.java:2365)
                                                                        at com.cjt2325.cameralibrary.CameraInterface.doStartPreview(CameraInterface.java:371)
                                                                        at com.cjt2325.cameralibrary.KCameraView.cameraHasOpened(KCameraView.java:247)
                                                                        at com.cjt2325.cameralibrary.CameraInterface.doOpenCamera(CameraInterface.java:304)
                                                                        at com.cjt2325.cameralibrary.KCameraView$5.run(KCameraView.java:275)

kevin321happy, 2018-08-27:
使用两个JCameraView对象时会报错

yangjiechina, 2018-10-10:
还有这种操作？？？

closed 2019-01-16 by CJT2325