CJT2325/CameraView #214: 找不到JCameraLisenter

handsonpig, 2019-02-14:
//JCameraView监听
        jCameraView.setJCameraLisenter(new JCameraLisenter() {
            @Override
            public void captureSuccess(Bitmap bitmap) {
                //获取图片bitmap
                Log.i("JCameraView", "bitmap = " + bitmap.getWidth());
            }

            @Override
            public void recordSuccess(String url, Bitmap firstFrame) {

            }

            @Override
            public void quit() {
                //退出按钮
                GetVedioActivity.this.finish();
            }
        });

这一句中new JCameraLisenter()依赖不上，找不到JCameraLisenter

ziqianghe, 2019-03-15:
同上 一样找不到

bestchenahua, 2019-03-19:
删掉 new JCameraLisenter 重新写一个 应该就有提示了

AlanTang2416, 2019-11-19:
这个接口的名字改了