CJT2325/CameraView #219: 版本代码是最新吗

muchmeat, 2019-03-19:


muchmeat, 2019-03-21:
发现一个问题，6.0及以上首次获取权限时页面黑屏。安装apk后去初次调用此功能时，页面会黑屏，退出再进入就正常了，这个问题能看一下吗

ZengCS, 2019-09-11:
我遇到同样的问题，我在外面申请完权限后，再打开CameraActivity

Kevin20145, 2019-11-26:
我的解决如下，大家可以参考一下：
1、JCameraView中surfaceCreated中如下：
@Override
    public void surfaceCreated(SurfaceHolder holder) {
        LogUtil.i("JCameraView SurfaceCreated");
        openCamera();
    }

    public void openCamera() {
        new Thread() {
            @Override
            public void run() {
                CameraInterface.getInstance().doOpenCamera(JCameraView.this);
            }
        }.start();
    }
2、权限申请回调中，onRequestPermissionsResult中，增加 jCameraView.openCamera();，修改如下：
if (size == 0) {
                    granted = true;
                    jCameraView.openCamera();
                } else {
                    Toast.makeText(this, "请到设置-权限管理中开启", Toast.LENGTH_SHORT).show();
                    finish();
                }

FangHongDa, 2020-01-13:
可以申请完权限后 重启 activity
Intent intent = getIntent();
    overridePendingTransition(0, 0);
    intent.addFlags(Intent.FLAG_ACTIVITY_NO_ANIMATION);
    finish();
    overridePendingTransition(0, 0);
    startActivity(intent);