CJT2325/CameraView #223: 全面屏手机预览时，画面拉伸了 18：9比例

tenda2014, 2019-04-02:
有个方法，重新绘制下布局，这个是我给整合得，我试了下可以用。希望楼主能看看。
` 
@Override
    public void cameraHasOpened() {
        CameraInterface.getInstance().doStartPreview(mVideoView.getHolder(), screenProp);
        setSuitableParams();
    }
  private void setSuitableParams() {
        if (screenProp > 1.8) { //屏幕的宽高比大于1.8的时候处理
            float previewProp = CameraInterface.getInstance().getPrevewProp();
            if (previewProp == 0 || mVideoView == null) {   //获取的size宽高比
                return;
            }
            int measuredHeight = mVideoView.getMeasuredHeight();
            int measuredWidth = mVideoView.getMeasuredWidth();
            float clacWidth = measuredHeight / previewProp;   //计算出要显示的预览界面的宽度。
            ViewGroup.LayoutParams layoutParams = mVideoView.getLayoutParams();
            if (layoutParams == null) {
                layoutParams = new ViewGroup.LayoutParams((int) clacWidth, measuredHeight);
            }
//            if (clacWidth > 800 && Math.abs(clacWidth - measuredWidth) > clacWidth * 0.1F) {  //计算的宽度大于 800 并且和显示正常的布局的误差超过10%
            layoutParams.width = (int) clacWidth;
//            }
            final ViewGroup.LayoutParams finalLayoutParams = layoutParams;
            mVideoView.post(new Runnable() {
                @Override
                public void run() {
                    if (mVideoView != null && finalLayoutParams != null) {
                        mVideoView.setLayoutParams(finalLayoutParams);
                    }
                }
            });
        }
    }
`
CameraInterface 中doStartPreview方法增加获取分辨率比率值：
prevewProp = previewSize.width / previewSize.height;

zhuoshirong, 2019-04-10:
还是不行，OPPO手机还是变形

zhuoshirong, 2019-04-10:
如果照你这样的优化三得S9严重变形，预览图形

zhuoshirong, 2019-04-10:
如果照你这样的优化三星S9严重变形，预览图形

tenda2014, 2019-05-20:
> 如果照你这样的优化三星S9严重变形，预览图形

不好意思，有个地方改下
 prevewProp = (float) previewSize.width / (float) previewSize.height;
这个地方需要转成float类型。

XuHX, 2019-06-21:
9.0手机还是变形

XuHX, 2019-06-24:
主要是预览和图片分辨率不匹配,previewSize的分辨率是最高的，然而部分手机pictureSize又是最低的，导致预览的图片十分清晰，而拍出的照片十分小且模糊,两种解决方法,一将预览的图像保存为图片,二 根据得到的分辨率,设置预览分辨率和图片的分辨率一样或者值相近的,因为手机摄像头支持的参数可能不一样

xionger0520, 2019-12-22:
看下我重写的这个库  没有拉伸了 @tenda2014 @zhuoshirong @XuHX