CJT2325/CameraView #224: 大神们，享问下，相机启动时，是不是没有自动对焦了

andoria, 2019-04-15:
FirstFoucsLisenter 这个类都不存在了，求回复！谢谢

KevinSgun, 2019-04-17:
感觉除了第一次打开焦点是对的，后面你快速移动手机或拉远拉近距离都没有再自动对焦了

AzusaL, 2019-05-08:
对焦模式设置改成下面这段代码

 boolean isCanAutoFocus = false;
                if ((SELECTED_CAMERA == CAMERA_FRONT_POSITION && mParams.isSmoothZoomSupported()) ||
                        (SELECTED_CAMERA == CAMERA_POST_POSITION &&
                                (mParams.isSmoothZoomSupported() || mParams.isZoomSupported()))) {
                    isCanAutoFocus = true;
                }
                if (isCanAutoFocus) {
                    //设置对焦模式
                    if (mParams.getSupportedFocusModes().contains(Camera.Parameters.FOCUS_MODE_CONTINUOUS_VIDEO)) {
                        mParams.setFocusMode(Camera.Parameters.FOCUS_MODE_CONTINUOUS_VIDEO);
                    } else {
                        mParams.setFocusMode(Camera.Parameters.FLASH_MODE_AUTO);
                    }
                }