CJT2325/CameraView #231: 画面拉伸

zgh07280510, 2019-05-09:
在全面屏下,画面拉伸.

NickNian0522, 2019-11-22:
画面拉伸问题已解决！
思路如下：
 List<Camera.Size> sameSize = CameraParamUtil.getInstance()
                        .getTheSameSection(mParams.getSupportedPreviewSizes(),mParams.getSupportedPictureSizes());
                boolean hasSameSize = false;
                if(!sameSize.isEmpty()){
                    for (Camera.Size s : sameSize) {
                        if ((s.width > 1000) ) {
                            hasSameSize = true;
                            break;
                        }
                    }
                }
                Camera.Size previewSize = CameraParamUtil.getInstance().getPreviewSize
                        (hasSameSize?sameSize:mParams.getSupportedPreviewSizes(), 1000, screenProp);
                Camera.Size pictureSize = CameraParamUtil.getInstance().getPictureSize
                        (hasSameSize?sameSize:mParams.getSupportedPictureSizes(), 1000, screenProp);

xionger0520, 2019-12-22:
整得这么麻烦  我用camerax重写了这个库  不存在拉伸了  @zgh07280510