CJT2325/CameraView #232: 现在时长按录制，怎么样可以改成点击录制呢？

HopeString, 2019-05-13:


WhySRZ, 2019-05-14:
下源码自改一下就好了，应该很简单

ikakaxi, 2019-07-31:
这个库2年不更新了，但是你说的这个问题我解决了，我是设置了只录像，但是他这个库只录像的话有bug，你需要这样改
在CaptureButton类里增加这样的属性
`private Handler handler = new Handler() {
		@Override
		public void handleMessage(Message msg) {
			longPressRunnable.run();
		}
	};
`
然后修改onTouchEvent方法，将下面的代码
`
if ((button_state == BUTTON_STATE_ONLY_RECORDER || button_state == BUTTON_STATE_BOTH)) {
					postDelayed(longPressRunnable, 500);    //同时延长500启动长按后处理的逻辑Runnable
				}
`
改为
`
if ((button_state == BUTTON_STATE_ONLY_RECORDER || button_state == BUTTON_STATE_BOTH)) {
					handler.postDelayed(longPressRunnable, 500);    //同时延长500启动长按后处理的逻辑Runnable
				}
`

guoq1, 2019-08-01:
按照上面的代码修改后在补充一下,  点击录制后, 操作返回键, 录制完成自动播放的时候会崩
修改:
PreviewState.class
stopRecord()内的方法加一层if判断, 自己写一个getSurfaceHolder() {return mHolder;}
if (CameraInterface.getInstance().getSurfaceHolder() != null) {
                        machine.getView().playVideo(firstFrame, url);
                        machine.setState(machine.getBorrowVideoState());
                    }

ikakaxi, 2019-08-22:
楼上方法是在CameraInterface.java里面增加方法
```
	public SurfaceHolder getSurfaceHolder() {
		return mHolder;
	}
```
然后修改PreviewState.java类的stopRecord方法为
```
	@Override
	public void stopRecord(final boolean isShort, long time) {
		CameraInterface.getInstance().stopRecord(isShort, new CameraInterface.StopRecordCallback() {
			@Override
			public void recordResult(String url, Bitmap firstFrame) {
				if (isShort) {
					machine.getView().resetState(JCameraView.TYPE_SHORT);
				} else {
					if (CameraInterface.getInstance().getSurfaceHolder() != null) {
						machine.getView().playVideo(firstFrame, url);
						machine.setState(machine.getBorrowVideoState());
					}
				}
			}
		});
	}
```