CJT2325/CameraView #254: 小米手机崩溃

WarriorYu, 2019-11-12:
使用BUTTON_STATE_ONLY_RECORDER只录制视频功能，当点击录制按钮多次或者点击屏幕再点击录制按钮的时候，直接崩溃。
手机型号：Redmi Note 4X;Android 版本：7.0

com.zxing.cameraapplication E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.zxing.cameraapplication, PID: 26870
    java.lang.RuntimeException: getParameters failed (empty parameters)
        at android.hardware.Camera.native_getParameters(Native Method)
        at android.hardware.Camera.getParameters(Camera.java:2011)
        at com.cjt2325.cameralibrary.CameraInterface.doStartPreview(CameraInterface.java:368)
        at com.cjt2325.cameralibrary.state.BorrowVideoState.start(BorrowVideoState.java:28)
        at com.cjt2325.cameralibrary.state.CameraMachine.start(CameraMachine.java:71)
        at com.cjt2325.cameralibrary.JCameraView.confirmState(JCameraView.java:450)
        at com.cjt2325.cameralibrary.state.BorrowVideoState.confirm(BorrowVideoState.java:76)
        at com.cjt2325.cameralibrary.state.CameraMachine.confirm(CameraMachine.java:116)
        at com.cjt2325.cameralibrary.JCameraView$4.confirm(JCameraView.java:237)
        at com.cjt2325.cameralibrary.CaptureLayout$4.onClick(CaptureLayout.java:222)
        at android.view.View.performClick(View.java:5619)
        at android.view.View$PerformClick.run(View.java:22295)
        at android.os.Handler.handleCallback(Handler.java:754)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:163)
        at android.app.ActivityThread.main(ActivityThread.java:6342)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:880)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:770)

xionger0520, 2019-12-22:
@WarriorYu  我测试的时候还有调用录像的时候动画错乱的问题  我自己用camerax重写了关于相机操作的部分  修复了大部分bug

lly7690, 2020-11-03:
@xionger0520  开源分享下？