CJT2325/CameraView #257: 华为、红米、Vivo机子上都出现过了，大佬有空解决下

opensoftwareS, 2019-12-10:
android.hardware.Camera.native_takePicture(Native Method)
--
2 android.hardware.Camera.takePicture(Camera.java:1623)
3 android.hardware.Camera.takePicture(Camera.java:1565)
4 com.cjt2325.cameralibrary.CameraInterface.takePicture(CameraInterface.java:469)
5 com.cjt2325.cameralibrary.state.PreviewState.capture(PreviewState.java:59)
6 com.cjt2325.cameralibrary.state.CameraMachine.capture(CameraMachine.java:96)
7 com.cjt2325.cameralibrary.JCameraView$3.takePictures(JCameraView.java:187)
8 com.cjt2325.cameralibrary.CaptureLayout$2.takePictures(CaptureLayout.java:151)
9 com.cjt2325.cameralibrary.CaptureButton$2.onAnimationEnd(CaptureButton.java:237)

xionger0520, 2019-12-22:
@opensoftwareS 这个问题无解  兼容性问题  就是由于国内大厂关于相机操作的改动有点多  问题自然多  作者感觉不更新这个项目了   我已经用camerax重写了关于相机操作的部分

opensoftwareS, 2020-01-14:
好的我去看一下camera相关内容去，谢谢你

opensoftwareS, 2020-01-14:
好的我去看一下camerax相关内容去，谢谢你

xionger0520, 2020-01-14:
@opensoftwareS  看下我的库基于这个重写  适配了Android Q

opensoftwareS, 2020-01-14:
好的谢谢大佬

timfei, 2020-04-21:
你好，请问最后有解决改问题吗？我也遇到这样的报错