CJT2325/CameraView #261: 引入源码，修改vedio名字，录制视频成功后导致崩溃

xuzhu19930323, 2020-01-08:
崩溃代码：
 
//videoFileName = "video_" + System.currentTimeMillis() + ".mp4";
        SimpleDateFormat sdf = new SimpleDateFormat("yyyy-MM-dd HH:mm", Locale.CHINA);
        String date = sdf.format(System.currentTimeMillis());
        videoFileName = date + "_" + "10174" + ".mp4";

崩溃信息：

java.lang.RuntimeException: startPreview failed
        at android.hardware.Camera.startPreview(Native Method)
        at com.iqizu.user.camera.CameraInterface.doStartPreview(CameraInterface.java:400)
        at com.iqizu.user.camera.state.BorrowVideoState.start(BorrowVideoState.java:28)
        at com.iqizu.user.camera.state.CameraMachine.start(CameraMachine.java:72)
        at com.iqizu.user.camera.JCameraView.confirmState(JCameraView.java:456)
        at com.iqizu.user.camera.state.BorrowVideoState.confirm(BorrowVideoState.java:76)
        at com.iqizu.user.camera.state.CameraMachine.confirm(CameraMachine.java:117)
        at com.iqizu.user.camera.JCameraView$4.confirm(JCameraView.java:239)
        at com.iqizu.user.camera.CaptureLayout$4.onClick(CaptureLayout.java:222)
        at android.view.View.performClick(View.java:6329)
        at android.view.View$PerformClick.run(View.java:25002)
        at android.os.Handler.handleCallback(Handler.java:809)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:166)

pumatus, 2020-10-09:
有找到解决方案吗？

lly7690, 2020-11-03:
这个怎么解决啊？startpreview failed

pumatus, 2020-11-04:
安安稳稳的设置flag吧