CJT2325/CameraView #266: VIVO X21 UDA  MediaRecorder: start failed: -38

daibao520, 2020-03-31:
V/Camera: isNormalCamera packageName = com.zxing.cameraapplication
    isTopActivity
I/Camera: isTopActivity: pkgname: com.zxing.cameraapplication
V/Camera: isNormalCamera isTopActivity true
    sendOpenCameraBroadCast isNormalCamera = true
I/JCameraView: MakeSure Preview :w = 2280 h = 1080
I/JCameraView: MakeSure Picture :w = 5632 h = 2816
I/JCameraView: FocusMode supported auto
I/JCameraView: Formats supported 256
V/Camera: sendStartPreviewBroadCast
I/CJT: === Start Preview ===
I/CJT: state = 1
D/AudioRecord: AudioRecord::start
    client name com.zxing.cameraapplication(21371)
    misRecordUseJoviMic:0,misRecordUseMainMic:0
D/AudioRecord: AudioRecord::stop
D/skia: onFlyCompress
V/MediaRecorder: constructor
V/MediaRecorder: doCleanUp
    setListener
    setClientName
V/MediaRecorder: reset
    doCleanUp
    setCamera(0x7b9cbaa140,0x7b9cbaa400)
V/MediaRecorder: setVideoSource(1)
    Call init() since the media recorder is not initialized yet
    init
V/MediaRecorder: setAudioSource(1)
V/MediaRecorder: setOutputFormat(2)
V/MediaRecorder: setVideoEncoder(2)
V/MediaRecorder: setAudioEncoder(3)
I/CJT: setVideoSize    width = 864height = 480
V/MediaRecorder: setVideoSize(864, 480)
V/MediaRecorder: setParameters(video-param-rotation-angle-degrees=90)
V/MediaRecorder: setParameters(video-param-encoding-bitrate=1600000)
V/MediaRecorder: setOutputFile(65)
I/MediaRecorderJNI: prepare: surface=0x7b9cb48000
V/MediaRecorder: setPreviewSurface(0x7b9cb93e40)
V/MediaRecorder: prepare
V/MediaRecorder: start
E/MediaRecorder: start failed: -38
W/System.err: java.lang.IllegalStateException
W/System.err:     at android.media.MediaRecorder.start(Native Method)
W/System.err:     at com.cjt2325.cameralibrary.CameraInterface.startRecord(CameraInterface.java:621)
W/System.err:     at com.cjt2325.cameralibrary.state.PreviewState.record(PreviewState.java:71)
W/System.err:     at com.cjt2325.cameralibrary.state.CameraMachine.record(CameraMachine.java:101)
W/System.err:     at com.cjt2325.cameralibrary.JCameraView$3.recordStart(JCameraView.java:194)
        at com.cjt2325.cameralibrary.CaptureLayout$2.recordStart(CaptureLayout.java:166)
W/System.err:     at com.cjt2325.cameralibrary.CaptureButton$5.onAnimationEnd(CaptureButton.java:274)
W/System.err:     at android.animation.Animator$AnimatorListener.onAnimationEnd(Animator.java:552)
W/System.err:     at android.animation.AnimatorSet.endAnimation(AnimatorSet.java:1294)
        at android.animation.AnimatorSet.doAnimationFrame(AnimatorSet.java:1079)
W/System.err:     at android.animation.AnimationHandler.doAnimationFrame(AnimationHandler.java:146)
W/System.err:     at android.animation.AnimationHandler.access$100(AnimationHandler.java:37)
W/System.err:     at android.animation.AnimationHandler$1.doFrame(AnimationHandler.java:54)
W/System.err:     at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1017)
        at android.view.Choreographer.doCallbacks(Choreographer.java:831)
W/System.err:     at android.view.Choreographer.doFrame(Choreographer.java:760)
W/System.err:     at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1005)
W/System.err:     at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
W/System.err:     at android.os.Looper.loop(Looper.java:224)
W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:7042)
        at java.lang.reflect.Method.invoke(Native Method)
W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:536)
W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:868)
I/CJT: startRecord IllegalStateException
    camera error
I/CJT: state = 4

daibao520, 2020-03-31:
系统版本 9
SDK版本 28

xionger0520, 2020-04-01:
@daibao520 你可以试试我的flowcamera

CuiZhaoHui, 2021-02-03:
> 系统版本 9
> SDK版本 28

确实是不行，尝试了下，应该是Camera.setParams有某些参数在这个机型上不支持，且开启录制前需要释放一下Camera并重新打开。

CuiZhaoHui, 2021-02-03:
> 系统版本 9
> SDK版本 28

相机重开，params重设。应该可以使用了。

xionger0520, 2021-02-03:
我最近很忙，先自个修改一下源码看看能不能解决

发自我的iPhone

dnwalter, 2021-09-30:
> 系统版本 9 SDK版本 28

CameraInterface类的doStartPreview里，把下面这行注释掉
`mParams.setPreviewSize(previewSize.width, previewSize.height);`

完美解决！请叫我雷锋。

dnwalter, 2021-09-30:
@daibao520 
CameraInterface类的doStartPreview里，把下面这行注释掉
`mParams.setPreviewSize(previewSize.width, previewSize.height);`

完美解决！请叫我雷锋。